espressif/arduino-esp32 #11866: ESP32 Dev Module Upload/Driver Issues After Windows 11 Upgrade

Rushify, 2025-09-22:
### Board

ESP32 Dev Module

### Device Description

ESP32 Development Board (ESP32-WROOM-32 + CP2102)

Core chip: ESP32-D0WDQ6 (dual-core 32-bit LX6, 240 MHz, Wi-Fi + Bluetooth)

Module: ESP32-WROOM-32 (8 MB embedded flash)

### Hardware Configuration

Hardware:
- W5500: CS->5, RST->4, SPI pins
- Keypad: Rows(32,33,25,26), Cols(27,14,13)  
- LCD: SDA->21, SCL->22
- Fingerprint: TX->17, RX->16

### Version

v3.0.0

### Type

Question

### IDE Name

Arduino IDE

### Operating System

Windows 11

### Flash frequency

No idea

### PSRAM enabled

yes

### Upload speed

115200

### Description

OS: Windows 11 (worked fine on Windows 10)
Arduino IDE: Latest
Version : v3.0.0

After upgrading to Windows 11, uploads often fail with:

Failed to connect to ESP32: No serial data received
Could not open COM port (busy or missing)

What I’ve Tried:
Manual BOOT+RESET sequence (sometimes works randomly)
Different COM ports (COM3 → COM10)
Updated CP210x drivers (latest from Silicon Labs)
Different cables/USB ports

Disabling/enabling COM port in Device Manager (temporary fix but still inconsisten doesnt work 100%)

Current Status:
Board is connected
Device Manager shows Silicon Labs CP210x USB to UART Bridge (COM3)
Uploads succeed only inconsistently (rarely upload)
When upload succeeds, code + Serial Monitor work perfectly
Often need to disable/enable the CP210x device to make it upload again (doesn't work 100%)

Worked 100% on Windows 10 with the same hardware

### Sketch

```cpp
/*
===============================================================================
                         SIMPLE ESP32 HARDWARE TEST
===============================================================================

Quick Test for:
✓ W5500 Ethernet Module - Does it connect?
✓ 3x4 Keypad - Do keys work?
✓ LCD Display - Can we show text?
✓ Fingerprint Sensor - Does it respond?

Hardware:
- W5500: CS->5, RST->4, SPI pins
- Keypad: Rows(32,33,25,26), Cols(27,14,13)  
- LCD: SDA->21, SCL->22
- Fingerprint: TX->17, RX->16

===============================================================================
*/

#include <SPI.h>
#include <Ethernet.h>
#include <Keypad.h>
#include <Wire.h>
#include <LiquidCrystal_I2C.h>
#include <Adafruit_Fingerprint.h>

// W5500 Configuration
#define W5500_CS_PIN    5
#define W5500_RST_PIN   4
byte mac[] = { 0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0xED };
IPAddress ip(192, 168, 100, 200);

// 3x4 Keypad Configuration
const byte ROWS = 4, COLS = 3;
char keys[ROWS][COLS] = {
  {'1', '2', '3'},
  {'4', '5', '6'},
  {'7', '8', '9'},
  {'*', '0', '#'}
};
byte rowPins[ROWS] = {32, 33, 25, 26};
byte colPins[COLS] = {27, 14, 13};
Keypad keypad = Keypad(makeKeymap(keys), rowPins, colPins, ROWS, COLS);

// LCD Configuration
LiquidCrystal_I2C* lcd = nullptr;
bool lcdWorking = false;

// Fingerprint Configuration
HardwareSerial fingerprintSerial(1);
Adafruit_Fingerprint finger = Adafruit_Fingerprint(&fingerprintSerial);

void setup() {
  Serial.begin(115200);
  delay(2000);
  
  Serial.println("=== ESP32 SERIAL TEST ===");
  Serial.println("ESP32 is alive and starting hardware tests!");
  
  // 1. Test LCD
  Serial.print("Testing LCD... ");
  if (testLCD()) {
    Serial.println("✓ LCD Working!");
    lcd->clear();
    lcd->setCursor(0, 0);
    lcd->print("Hardware Test");
    lcd->setCursor(0, 1);
    lcd->print("Starting...");
  } else {
    Serial.println("✗ LCD Failed");
  }
  
  delay(2000);
  
  // 2. Test Fingerprint Sensor
  Serial.print("Testing Fingerprint Sensor... ");
  if (testFingerprint()) {
    Serial.println("✓ Fingerprint Working!");
    if (lcdWorking) {
      lcd->clear();
      lcd->setCursor(0, 0);
      lcd->print("Fingerprint OK");
      lcd->setCursor(0, 1);
      lcd->print("Sensor Ready");
    }
  } else {
    Serial.println("✗ Fingerprint Failed");
    if (lcdWorking) {
      lcd->clear();
      lcd->setCursor(0, 0);
      lcd->print("Fingerprint ERR");
      lcd->setCursor(0, 1);
      lcd->print("Check wiring");
    }
  }
  
  delay(2000);
  
  // 3. Test W5500
  Serial.print("Testing W5500... ");
  if (testW5500()) {
    Serial.println("✓ W5500 Working!");
    if (lcdWorking) {
      lcd->clear();
      lcd->setCursor(0, 0);
      lcd->print("W5500 OK");
      lcd->setCursor(0, 1);
      lcd->print("Ethernet Ready");
    }
  } else {
    Serial.println("✗ W5500 Failed");
    if (lcdWorking) {
      lcd->clear();
      lcd->setCursor(0, 0);
      lcd->print("W5500 ERROR");
      lcd->setCursor(0, 1);
      lcd->print("Check wiring");
    }
  }
  
  delay(2000);
  
  // 4. Ready for keypad test
  Serial.println("Press keys to test keypad...");
  if (lcdWorking) {
    lcd->clear();
    lcd->setCursor(0, 0);
    lcd->print("Press Keys");
    lcd->setCursor(0, 1);
    lcd->print("Test Keypad");
  }
}

bool testLCD() {
  Wire.begin(21, 22); // SDA, SCL
  
  // Try common LCD addresses
  uint8_t addresses[] = {0x27, 0x3F, 0x26, 0x20};
  
  for (int i = 0; i < 4; i++) {
    Wire.beginTransmission(addresses[i]);
    if (Wire.endTransmission() == 0) {
      lcd = new LiquidCrystal_I2C(addresses[i], 16, 2);
      lcd->init();
      lcd->backlight();
      lcdWorking = true;
      return true;
    }
  }
  return false;
}

bool testFingerprint() {
  // Initialize fingerprint sensor
  fingerprintSerial.begin(57600, SERIAL_8N1, 16, 17);
  finger.begin(57600);
  
  delay(500);
  
  // Try to verify fingerprint sensor
  if (finger.verifyPassword()) {
    return true;
  }
  return false;
}

bool testW5500() {
  // Reset W5500
  pinMode(W5500_RST_PIN, OUTPUT);
  digitalWrite(W5500_RST_PIN, LOW);
  delay(100);
  digitalWrite(W5500_RST_PIN, HIGH);
  delay(500);
  
  // Initialize Ethernet
  Ethernet.init(W5500_CS_PIN);
  Ethernet.begin(mac, ip);
  delay(1000);
  
  // Check if W5500 is detected
  return (Ethernet.hardwareStatus() != EthernetNoHardware);
}

void loop() {
  // Keep alive message
  static unsigned long lastAlive = 0;
  if (millis() - lastAlive > 5000) {
    Serial.println("ESP32 alive - counter: " + String(millis()));
    lastAlive = millis();
  }
  
  // Test keypad
  char key = keypad.getKey();
  
  if (key) {
    Serial.print("Key pressed: ");
    Serial.println(key);
    
    if (lcdWorking) {
      lcd->clear();
      lcd->setCursor(0, 0);
      lcd->print("Key Pressed:");
      lcd->setCursor(0, 1);
      lcd->print(key);
    }
    
    delay(300); // Debounce
  }
}
```

### Debug Message

```plain
A fatal error occurred: Failed to connect to ESP32: No serial data received.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

TD-er, 2025-09-22:
Is this happening after the latest update, or did you do a massive 'jump' from Windows 10 to 11?
The reason I'm asking is because I had to roll-back the latest Windows 11 update about a week ago as I could no longer get USB to serial devices mapped to WSL2.
I figured it might be something in WSL2, but given this issue, it might be related to Windows itself?

SuGlider, 2025-09-22:
@Rushify - In order to verify the ESP32 board, please check a few things:
1- Open "Device Manager" application with the ESP32 connected to the PC. It should display a COM port associated with the CP21xx UART to USB port.
2- In case that the Device Manager APP shows the ESP32 COM port, please install/use some other Serial Terminal application such as PuTTY. Connect and open the COM port and check if the ESP32 Boot Messages can be viewed with 115200 baudrate.

Let me know the results.
Thanks.

Rushify, 2025-09-22:
> Is this happening after the latest update, or did you do a massive 'jump' from Windows 10 to 11? The reason I'm asking is because I had to roll-back the latest Windows 11 update about a week ago as [...]

I didn’t update from one recent Windows 11 build to another — I made a direct upgrade “jump” from Windows 10 → Windows 11.

On Windows 10 the ESP32 + CP210x driver worked 100% reliably.
After upgrading to Windows 11 the USB-to-serial connection became inconsistent (upload fails / COM port busy or missing).

So in my case, it’s not a recent patch issue but something that only started after moving to Windows 11.

Rushify, 2025-09-22:
> [@Rushify](https://github.com/Rushify) - In order to verify the ESP32 board, please check a few things: 1- Open "Device Manager" application with the ESP32 connected to the PC. It should display a [...]

Yes, I already checked that:

In Device Manager, the ESP32 shows up as “Silicon Labs CP210x USB to UART Bridge (COM10)” whenever it’s plugged in.

I also tested with PuTTY at 115200 baud. Sometimes I can see the normal ESP32 boot messages, but other times the COM port is busy or unresponsive until I disable/enable the CP210x device in Device Manager.

So the port is detected, but the behavior is inconsistent — which makes me think it’s related to Windows 11’s handling of the CP210x driver.

TD-er, 2025-09-25:
Great!
Good to get some confirmation that the latest drivers are OK (... again).

Rushify, 2025-09-27:
> Fantastic [@TD-er](https://github.com/TD-er)! Excellent!! Thanks for taking time to help!
> 
> [@Rushify](https://github.com/Rushify) - Do you consider this issue solved?

Yes I do