espressif/arduino-esp32 #11881: show IRK in BLE client sketch

bartgrefte, 2025-09-28:
### Related area

Bluetooth

### Hardware specification

ESP32 and Nothing Watch Pro 2

### Is your feature request related to a problem?

There are various topics about the subject, https://github.com/agittins/bermuda/discussions/551 , https://github.com/agittins/bermuda/wiki/How-to-get-the-IRK-(Identity-Resolving-Keys)-for-iOS,-Android-etc and on the Home Assistant forum 
https://community.home-assistant.io/t/guide-getting-the-irk-for-your-phone/623694 , 
https://community.home-assistant.io/t/struggling-to-find-an-irk-key-recognisable-by-private-ble-device-integration/875592/ , 
https://community.home-assistant.io/t/private-ble-device-for-use-with-bermuda-ble-trilateration-and-others-i-guess/778831 , 
https://community.home-assistant.io/t/bermuda-bluetooth-ble-room-presence-and-tracking-custom-integration/625780

### Describe the solution you'd like

Home Assistant has BLE-based presence detection, in order to connect my phone and smartwatch, I need the BLE Identity Resolving Key (IRK). I was able to obtain the one from my phone using https://github.com/fryefryefrye/Decoding-Random-Bluetooth-Address/tree/master/esp32_get_irk/Arduino/esp32_get_irk , but that solution requires a device to connect to the ESP32's bluetooth, my smartwatch can't do that, it needs to be connected to.

So what I need is a way to have https://github.com/espressif/arduino-esp32/tree/master/libraries/BLE/examples/Client to show the IRK like the other mentioned sketch does, only this while the ESP32 acts as a client instead of a server.

### Describe alternatives you've considered

On https://www.home-assistant.io/integrations/private_ble_device there are various ways mentioned, so far unsuccessful:
- the Android adb-method: No IRK of my smartwatch in the capture
- the Windows-registry method: No IRK on various systems, various Windows versions and 6 different BT-controllers. Also checked out Ubuntu, found some info about the connection, the IRK wasn't part of that info
- the earlier mentioned ESP32-method, which requires the device you want the IRK of to connect to the ESP32, the smartwatch can only be connected to.

### Additional context

The IRK is required for Home Assistant in order to do presence detection with certain BLE-devices that randomly change their bluetooth-address, see https://www.home-assistant.io/integrations/private_ble_device for more info.

### I have checked existing list of Feature requests and the Contribution Guide

- [x] I confirm I have checked existing list of Feature requests and Contribution Guide.

lucasssvaz, 2025-11-14:
Added in https://github.com/espressif/arduino-esp32/pull/11948

bartgrefte, 2025-11-14:
@lucasssvaz Thanks :) , when can I expect this release to become available in the Arduino IDE? It's on 3.3.3 there, this addition seems to be in 3.3.4?

bartgrefte, 2025-11-15:
@lucasssvaz I have https://dl.espressif.com/dl/package_esp32_index.json in the config (been there for years), "esp32 by Espressif Systems" is available up to 3.3.3, no update to 3.3.4 is picked up.

me-no-dev, 2025-11-15:
@bartgrefte please change the json link to one of the ones in the documentation that @lucasssvaz linked. Your link is years old and no longer supported. You get 3.3.3 because Arduino IDE also includes versions of our core