espressif/arduino-esp32 #11882: add RS485 Gateway to Example

DaTTcz, 2025-09-28:
### Related area

Example Code

### Hardware specification

ESP32-C6

### Is your feature request related to a problem?

Hi, I would like to ask for some sample code for a gateway between RS-485 and Zigbee.

### Describe the solution you'd like

I want to expand the Zigbee network using an RS-485 network. I will have several ESP32-C6 connected via RS-485. The first ESP32-C6/RS485 will be connected to the Zigbee coordinator via Zigbee. The coordinator will then use RS485 to connect to the second to Xth ESP32-C6/485, which will allow Zigbee devices to be registered with the coordinator. I hope I have described it clearly enough. Thank you.

### I have checked existing list of Feature requests and the Contribution Guide

- [x] I confirm I have checked existing list of Feature requests and Contribution Guide.

BitBrokerEU, 2025-09-30:
If you want to build a gateway I think you also need states of connected ZIGBEE EPs
This is a problem that I need to solve my Feature request
[How to get the state of a ZIGBEE EP i.e. a lamp](https://github.com/espressif/arduino-esp32/issues/11886)
Do you have a solution or is there already an example?

SuGlider, 2025-10-02:
>I want to expand the Zigbee network using an RS-485 network. 

I think that this is an specific user gateway application implementation. It is not a Zigbee feature.
There are many RS485 Arduino Libraries for ESP32 SoCs.
Is it going to use Modbus? Or some proprierary protocol?

The gateway shall implement both sides, with message/protocol translation.

This issue will be migrated to Discussions.