espressif/arduino-esp32 #11883: 3.3.1 release breaks USB Serial connection to Seeed Studio Xiao esp32C3

cattledogGH, 2025-09-28:
### Board

Seeed Studio Xiao ESP32C3

### Device Description

Seeed Studio Xiao ESP32C3

### Hardware Configuration

No external connections

### Version

v3.3.1

### Type

Bug

### IDE Name

Arduino IDE 1.8.19

### Operating System

Windows 11

### Flash frequency

80 MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

Tools > Port is greyed out.

Release 3.3.0 worked properly with IDE1.8.19 and showed connected usb port, but release 3.3.1 has port greyed out.
Both releases show the correct port with IDE 2.3.6

### Sketch

```cpp
Bare Minimum example sketch in Arduino IDE

void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}
```

### Debug Message

```plain
No error message
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

jameszah, 2025-09-28:
Specific to Arduino 1.8.19 - You could use Arduino 2.xx

https://github.com/espressif/esp32-camera/issues/775#issuecomment-3325376951

cattledogGH, 2025-09-29:
I believe that I have experienced the issue described here

boards.txt file now too big for Arduino IDE v1.8.19 and breaks the "Tools/Port" COM-Port access. #11798

[https://github.com/espressif/arduino-esp32/issues/11798](url)

3.3.0 boards.txt 2518kb
3.3.1 boards.txt 2595kb 

Removing some boards from the list has solved the issue and a modified boards.txt file of 2550kb does not cause the greying out of the available port.

SuGlider, 2025-09-29:
@cattledogGH 
Issue confirmed with Win11.
Reason seems to be related to https://github.com/espressif/arduino-esp32/issues/11798
Too many boards in the `Boards.txt` file --> too big for Arduino IDE 1.8.9 ?!

It works fine with Arduino IDE 2.3.6.

Not sure if ESP32 Arduino Core would be able to solve it... Maybe a clean up?!
FYI. @me-no-dev