espressif/arduino-esp32 #11884: Esreessif 32 fails to connect to Wifi if Asus SmartConnect is enabled.  errno: 118, "Host is unreachable"

AndKe, 2025-09-28:
### Board

LilyGo TT32 LoRa32

### Device Description

After switching on ASUS Smart Switch on the router(and accesspoint)  everything works great and roams fine,  but this one device fails to connect.

I get:
` 12017][E][WiFiClient.cpp:249] connect(): connect on fd 48, errno: 118, "Host is unreachable"`

I am using 
#include <WiFi.h>
and  connect using
WiFi.begin(ssid, password);

### Hardware Configuration

N/A

### Version

latest stable Release (if not listed below)

### Type

Task

### IDE Name

VSCODE

### Operating System

Ubuntu 25.04

### Flash frequency

40Mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

Please note that the library in VSCode / Platformio may be old?
```
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/ttgo-lora32-v21.html
PLATFORM: Espressif 32 (6.11.0) > TTGO LoRa32-OLED v2.1.6
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20017.0 (2.0.17) 
 - tool-esptoolpy @ 1.40501.0 (4.5.1) 
 - tool-mkfatfs @ 2.0.1 
 - tool-mklittlefs @ 1.203.210628 (2.3) 
 - tool-mkspiffs @ 2.230.0 (2.30) 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 38 compatible libraries
Scanning dependencies...
Dependency Graph
|-- LoRa @ 0.8.0
|-- Adafruit SSD1306 @ 2.5.13
|-- PubSubClient @ 2.8.0
|-- Adafruit GFX Library @ 1.11.10
|-- SPI @ 2.0.0
|-- WiFi @ 2.0.0
|-- Wire @ 2.0.0
Building in release mode
Retrieving maximum program size .pio/build/ttgo-lora32-v21/firmware.elf
Checking size .pio/build/ttgo-lora32-v21/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]  13.9% (used 45400 bytes from 327680 bytes)
Flash: [======    ]  60.2% (used 788773 bytes from 1310720 bytes)
Configuring upload protocol...
AVAILABLE: cmsis-dap, esp-bridge, esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = esptool
Looking for upload port...
Using manually specified: /dev/ttyACM0
Uploading .pio/build/ttgo-lora32-v21/firmware.bin
esptool.py v4.5.1
Serial port /dev/ttyACM0
Connecting.....
Chip is ESP32-PICO-D4 (revision v1.0)
Features: WiFi, BT, Dual Core, 240MHz, Embedded Flash, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: d4:d4:da:9e:a1:a4
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
```

### Sketch

```cpp
just basic wifi connect / the sketch is not the issue - unless it can do something differently / in addition to very basic connection.
```

### Debug Message

```plain
if I try to use platform version 6.12.0 I suddenly need python venv and a hex-library
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2025-09-29:
@AndKe Your version of this core is very very old. PlatformIO supports up to 2.x and we are currently on 3.3.1.  Please try with pioarduino

AndKe, 2025-09-29:
@me-no-dev please explain what this tells me to do "Please try with pioarduino"  
do you mean this: https://github.com/pioarduino/pioarduino-vscode-ide
I am surprised that PlatformIO does not ... maintain it beyond 2.x - is this a known situation?

AndKe, 2025-09-29:
Thank you, I'll close this issue now as this project is clearly not to blame.