espressif/arduino-esp32 #11885: V3.3.1 ArduinoOTA.cpp SHA2Builder.h not found

AcuarioCat, 2025-09-29:
### Board

ESP32S3

### Device Description

ESP32S3

### Hardware Configuration

No

### Version

v3.3.1

### Type

Bug

### IDE Name

Visual Micro

### Operating System

Windows

### Flash frequency

40

### PSRAM enabled

yes

### Upload speed

921500

### Description

The V3.3.1 has updated the ArduinoOTA.cpp code and the SHA2Builder.h header is not found when compiling.
I have not checked to see if the PBKDF2_HMACBuilder.h header is also missing.

### Sketch

```cpp
#include <ArduinoOTA.h>
```

### Debug Message

```plain
None
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

AcuarioCat, 2025-09-29:
Investigating a bit further I found the headers in the hash directory of the libraries folder.
I don´t know why they are not automatically picked up by the compiler. 
The project compiles after adding the hash library to the sketch.
Any reason why it is not automatically picked up?

AcuarioCat, 2025-10-19:
Still happens with 3.3.2

hmccurryWM, 2025-10-19:
FYI I had this very same issue when I was mucking about with excluding arduino components with the ESP-IDF's configuration options. If you are using the core inside of ESP-IDF, then check that `KCONFIG Name: ARDUINO_SELECTIVE_COMPILATION` is not enabled. If you aren't using that, then 🤷‍♀️