espressif/arduino-esp32 #11886: How to get the state of a ZIGBEE EP i.e. a lamp

BitBrokerEU, 2025-09-29:
### Related area

Is a state callback for EPs available or possible?

### Hardware specification

ESP32-C6 Dev Kit

### Is your feature request related to a problem?

I build a light control on my ESP32-C6 Dev Kit which works very well.
Based on the example Zigbee_On_Off_MultiSwitch from Jan Procházka (/P-R-O-C-H-Y/).

I want to modify light intensity or if possible also light color relativ to the current state.
But how to get the state of i.e. a lamp?

### Describe the solution you'd like

I don't found an (simple) example to create a callback for a given EP or to get the state.
Did I not understand something or is this not possible?

### I have checked existing list of Feature requests and the Contribution Guide

- [x] I confirm I have checked existing list of Feature requests and Contribution Guide.

P-R-O-C-H-Y, 2025-09-29:
@BitBrokerEU I think you may need to use the [Color Dimmer Switch example](https://github.com/espressif/arduino-esp32/tree/master/libraries/Zigbee/examples/Zigbee_Color_Dimmer_Switch) now to have the commands to control light level and color.

BitBrokerEU, 2025-09-29:
Hi Jan, I use parts of them and I wrote it works very well ;-) Thanks for your examples!!!
But I need to adjust a lamps level (and color) where a user change something and the ESP32 should correct it.
Example: User set level to 50% then my ESP should adjust the colors to 100, 128, 156
Set the User to level 75% it should adjust the colors to i.e. 175, 200, 255 and so on...
So I need a trigger (or I made a endless loop) to get changes to the level and than I make color adjusts.
The question is: How to get (smart) changes of the level of a lamp or ...?

BitBrokerEU, 2025-09-29:
That's clear, but how can I connect to this function of the extisting EPs?

I explain the problem and hope then it will be clear:
The lamps are i.e. IKEA TRÅDFRI or other, dimmable and with colors.
These lamps will switch on or off from different buttons / switches / dimmer like IKEA Dirigera or others by different useractions.

My ESP-C6 ZIGBEE controller should react to these state changes (of the lamps)!
So, i.e. if lamp A and B are switched on, lamp C should turn on dimming plus 20% and switch to green color.
So I need the state switch of lamp A and B and the dimmerstate of lamp C to react...

I create witch the [Color Dimmer Switch example](https://github.com/espressif/arduino-esp32/tree/master/libraries/Zigbee/examples/Zigbee_Color_Dimmer_Switch) a Switch EP to manage the commands for the lamps.
That's easy with the examples from Jan Procházka (/P-R-O-C-H-Y/).
But I don't get the state changes from the lamps or the currend dimmerstate.

So how can I connect to this lamps (to existing EPs) to define a callback or so?
Or maybe there is a hook function to read an existing EP?
I don't found a solution to attach...

BitBrokerEU, 2025-10-07:
@P-R-O-C-H-Y I am on a business trip and can test it earliest on the weekend.I will report ...

closed 2025-10-08 by me-no-dev